Frix-x/klippain-shaketune #101: asxl chip error running vibrations measurement

CD1010, 2024-05-04:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

3.0

### Describe the bug and expected behavior

Just got latest kippain tonight.  getting this error when i try to run machine vibrations measurement.

I was able to run belt tuning no issue, so adxl is fine.  I am using beacon as my adxl.

7:24 PM The value 'adxl345' is not valid for CHIP 7:24 PM The value 'adxl345' is not valid for CHIP 7:24 PM CoreXY kinematics mode 7:24 PM 7:24 PM This operation can not be interrupted by normal means. Hit the "emergency stop" button to stop it if needed 7:24 PM Starting machine vibrations profile measurement__

### Additional information and klippy.log

[klippy (1).zip](https://github.com/Frix-x/klippain-shaketune/files/15211246/klippy.1.zip)
[mcu adxl]
serial: /dev/serial/by-id/usb-Anchor_Rampon-if00

[adxl345]
cs_pin: adxl:CS

[resonance_tester]
accel_chip:  adxl345 adxl345
probe_points:
   150,150,40
# Edit the above line with the correct probe points location. I recommend the center of your bed for X & Y, 20 for Z.
# For 350 mm printers: 175,175,20
# For 300 mm printers: 150,150,20
# For 250 mm printers: 125,125,20
# For 120 mm printers: 60,60,20

Frix-x, 2024-05-12:
I think this is a Klipper configuration problem because you have two chips on the same line here:
```
accel_chip:  adxl345 adxl345
```

CD1010, 2024-05-13:
I put that in as an experiment.  That is a way to give the accel a name in klipper.  Anyway, i the second asdl1345  and still got the error.

FPS: 14
​
Send code...
8:03 PM
The value 'adxl345' is not valid for CHIP
8:03 PM
The value 'adxl345' is not valid for CHIP

CD1010, 2024-05-13:
Tried calling this macro also
ACCELEROMETER_MEASURE CHIP=adxl345  It complained with same error  ( I saw that create_vibrations_profile calls it ) 
Not familiar with this macro language.

Frix-x, 2024-05-13:
Yes, so that's definitely a Klipper problem then as ACCELEROMETER_MEASURE is a system command.
I think you should look for help on Klipper discord directly about it and look in https://www.klipper3d.org/Measuring_Resonances.html#software-installation or in the Beacon documentation then as I'm not really familiar with this device...

closed 2024-05-19 by Frix-x