Frix-x/klippain-shaketune #103: Are the tests supposed to be done in a certain location on the XY?

MackinsVII, 2024-05-08:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v3.0.0-1-g56a5502d

### Describe the bug and expected behavior

My bed size is 300x300, but when I run these tests, my toolhead goes to 110, 110 to perform the tests. 

Is this expected behaviour? Should it not do them at 150, 150 so the belts are even?

closed 2024-05-08 by MackinsVII

VectorForce3D, 2024-06-14:
What was the solution to this?

Frix-x, 2024-06-14:
> What was the solution to this?

The test is done at the [resonance_tester] probe_point site. So validate that it is set at the correct position here

VectorForce3D, 2024-06-14:
Yea this is what I forgot about when I set up my ADXL.cfg. I ended up finding it yesterday.