Frix-x/klippain-shaketune #107: Diffrent Results between Klippers Shaper Calibration and Klippain on X Axis

NightHammer1000, 2024-05-17:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v3.0.0

### Describe the bug and expected behavior

Running the Klippers own Shaper Calibration gives me widely different results then using Klippain.

Built In Suggests:

The Results for X is this:
[image]

While Klippain gives me this:
[image]

Somehow, the Peaks are flipped in Klippain. The Klipper built in Results gives much better Print Quality too.

Y, on the other hand, seems Identical. Even tho, Klippain suggests other settings than Built-in:
Built In:
[image]
Kliappain:
[image]

The printer is a QIDI X-PLUS set up from scratch with Upstream Klipper.

Frix-x, 2024-05-19:
Hello!

This seems a little strange to me. Are you sure it is reproducible?
Actually it's a bit strange to me because my macros call the original Klipper TEST_RESONANCE under the hood, so all the measurement part is done directly by Klipper. Shake&Tune is then based on the original plotting script, but with added insights. This means that there should be no difference in the recorded signal, only in the cosmetic elements added to it (like peak detection, etc...).

NightHammer1000, 2024-05-19:
Yep. It is reproducable. Again and again.
The Peaks are always flipped.

Did 3 runs right now. All 3 have the same result.

closed 2024-05-19 by NightHammer1000