Frix-x/klippain-shaketune #116: Updgrade to version 4.0.0 fails in RatRig V-Core 3.1 RatOS 2.0.2

qruret, 2024-06-13:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

4.0.0

### Describe the bug and expected behavior

Running wget -O - https://raw.githubusercontent.com/Frix-x/klippain-shaketune/main/install.sh | bash  failed to complete, leading in the following error: 

ERROR: Could not install packages due to an OSError: [Errno 13] Permission denied: '/home/pi/klippy-env/lib/python3.9/site-packages/mpl_toolkits/mplot3d/proj3d.py'
Check the permissions.

Running the same command with sudo in front of it did not help. After rebooting my raspberry now the firmware
does not start at all, stating Error: Internal error during connect: No module named 'matplotlib'

[klippy(2).log](https://github.com/user-attachments/files/15818906/klippy.2.log)

Frix-x, 2024-06-14:
Mhmm well, I think there is something wrong with how RatOS creates the Klipper virtual env. I'm not exactly sure why they're doing this, but the virtual env should not be created by the root user.

Can you check the permissions with: `ls -la ~/` ?

If the venv belongs to the root user, then this is something that RatOS needs to fix on their end, as I'm not sure why they need this and it could be a bug. In the meantime, doing `sudo chown -R pi:pi ~/klippy-env` might fix your problem, but I am not sure if this would break something else...

jhorn70, 2024-06-14:
drwxr-xr-x  6 pi   pi    4096 May 27 21:58 klippy-env

hapklaar, 2024-06-15:
Getting the same error as @jhorn70 on RatOS 2.0.2. Last lines during install are

```
Collecting smmap<6,>=3.0.1 (from gitdb<5,>=4.0.1->GitPython==3.1.41->-r /home/pi/klippain_shaketune/requirements.txt (line 1))
  Downloading https://www.piwheels.org/simple/smmap/smmap-5.0.1-py3-none-any.whl (24 kB)
Requirement already satisfied: zipp>=3.1.0 in ./klippy-env/lib/python3.9/site-packages (from importlib-resources>=3.2.0->matplotlib==3.8.2->-r /home/pi/klippain_shaketune/requirements.txt (line 2)) (3.15.0)
Requirement already satisfied: six>=1.5 in ./klippy-env/lib/python3.9/site-packages (from python-dateutil>=2.7->matplotlib==3.8.2->-r /home/pi/klippain_shaketune/requirements.txt (line 2)) (1.16.0)
Installing collected packages: smmap, numpy, scipy, PyWavelets, gitdb, matplotlib, GitPython
  Attempting uninstall: numpy
    Found existing installation: numpy 1.23.4
    Uninstalling numpy-1.23.4:
      Successfully uninstalled numpy-1.23.4
  Attempting uninstall: scipy
    Found existing installation: scipy 1.13.0
    Uninstalling scipy-1.13.0:
      Successfully uninstalled scipy-1.13.0
  Attempting uninstall: matplotlib
    Found existing installation: matplotlib 3.7.1
    Uninstalling matplotlib-3.7.1:
ERROR: Could not install packages due to an OSError: [Errno 13] Permission denied: '__init__.py'
Check the permissions.
```

klippy-env is owned by user pi

jhorn70, 2024-06-15:
It's either the matplotlib folder or one of it's dependencies.  I made sure all of them were owned by pi.  

sudo chown -R pi:pi /home/pi/klippy-env/lib/python3.9/site-packages/matplotlib
sudo chown -R pi:pi /home/pi/klippy-env/lib/python3.9/site-packages/mpl_toolkits

Your mileage may vary, but that fixed it for me and I am not sure if something else got hosed.

hapklaar, 2024-06-15:
That worked for me as well. I had to remove the matplotlib folders from /home/pi/klippy-env/lib/python3.9/site-packages/ as well, then reinstall back with `~/klippy-env/bin/pip install --force-reinstall --no-deps matplotlib==3.9.0`. Then installed using install.sh and all went well.

Frix-x, 2024-06-17:
Thanks for the feedback and glad you found a solution.

On my side I can't do more in S&T but feel like this should be told to the RatOS guys as it's probably something not wanted.

I'll close the issue, but feel free to reopen if you think there's more discussion needed

closed 2024-06-17 by Frix-x