Frix-x/klippain-shaketune #117: 4.0.0 Axis Shaper Calibration causes Move Queue Overflow

DerNinsaa, 2024-06-13:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v4.0.0

### Describe the bug and expected behavior

After updating to the newest version 4.0.0 following the breaking changes and installation guide, running the axis_shaper_calibration "macro" for all axis works only for the x-axis. 
After that, starting the y-axis causes the mcu to shutdown, throwing "move queue overflow"

This was repeatable five times, with a soft and hard reboot between, as well as increased cooling of the mcu.
Used mcu is the BTT CB1, on a Manta M8P Board.

Error Message:
`// Error while generating the graphs: Internal error processing accelerometer data [[ 327.915568 1608.800546 -148.041188 9474.636058]
// [ 327.915698 1838.629195 -296.082377 9326.594869]
// [ 327.915827 1608.800546 -74.020594 9474.636058]
// ...
// [ 328.084739 2068.457845 -296.082377 9548.656652]
// [ 328.084868 1991.848295 -74.020594 9400.615463]
// [ 328.084997 1838.629195 -74.020594 9400.615463]]
// None
15:32:46  // MCU 'mcu' shutdown: Move queue overflow
// Once the underlying issue is corrected, use the
// "FIRMWARE_RESTART" command to reset the firmware, reload the
// config, and restart the host software.
// Printer is shutdown
15:32:46  !! MCU 'mcu' shutdown: Move queue overflow`

Klippy Log is attached aswell.

### Additional information and klippy.log

[klippy.log](https://github.com/user-attachments/files/15821371/klippy.log)

DerNinsaa, 2024-06-13:
Running the macro seperately for both axis using:
`AXES_SHAPER_CALIBRATION FREQ_START=5 FREQ_END=133.33 HZ_PER_SEC=1 ACCEL_PER_HZ= AXIS='x' SCV= MAX_SMOOTHING= TRAVEL_SPEED=120 Z_HEIGHT=`
And 
`AXES_SHAPER_CALIBRATION FREQ_START=5 FREQ_END=133.33 HZ_PER_SEC=1 ACCEL_PER_HZ= AXIS='y' SCV= MAX_SMOOTHING= TRAVEL_SPEED=120 Z_HEIGHT=`
respectively is confimed to also not work when ran consecutively aswell.

Runnin the two commands with a `firmware_restart` between them does however produce the expected result

DerNinsaa, 2024-06-13:
Attached you find the "bugged" y axis graph made by shaketune when the shutdown occured.

[image]
[image]
[image]

MrQu33f, 2024-06-13:
After updating to the latest version I have also had the same issue. When performing y axis tuning I get an 'MCU 'mcu' shutdown: Move queue overflow' error also

Frix-x, 2024-06-14:
Thanks for reporting this. I'm aware of it but doesn't have a solution beside the Klipper restart in between for now.
I'm investigating

godbnc, 2024-06-14:
same problem here:) axis detected ->  z, -y, y   x not detected i have z, -y,x   mcu timer to close is shutdown only with AXES_SHAPER_CALIBRATION only with Y AXIS

EricZimmerman, 2024-06-16:
I had to disable autotune on .y k3 for y axis to finish

Frix-x, 2024-06-16:
This issue is highly linked to #120 
It is either a Move Queue overflow or Timer too close error that happens to people having loaded or low powered Pi clones. Usually running CANbus + lot of neopixels, etc... add to the probability of getting this error. I'm working on it.

MrQu33f, 2024-06-16:
For clarity, I am indeed using a BTT CB2 on my main board

EricZimmerman, 2024-06-16:
Pi5 here. Octo pro 723 1.1

godbnc, 2024-06-16:
No work with pi, btt pi i tried with orange pi but same result :)

Frix-x, 2024-06-16:
I pushed a potential fix on this branch: https://github.com/Frix-x/klippain-shaketune/tree/fix-timer-too-close
Would you mind testing it please and tell me if it works? Don't forget to restart the klipper service after git checkout the branch to be sure you are running the latest version of the code.

closed 2024-06-17 by Frix-x