Frix-x/klippain-shaketune #122: Timer too close when running any Shake and tune test

flavi0gritti, 2024-06-15:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v4.0.0-1-g8b0862a9

### Describe the bug and expected behavior

After updating to Shake and tune 4.0 this evening I can no longer run any of the test as I am constantly getting "timer too close" errors.
I have been running the same tests this morning on the same exact machine, no hardware changes, on version 3.0 I wasn't getting any crashes at all.
[image]

### Additional information and klippy.log

[klippy (1).log](https://github.com/user-attachments/files/15848189/klippy.1.log)

jhorn70, 2024-06-15:
I am getting this result as well on every test I run with 4.0

godbnc, 2024-06-15:
Ia global problem i think is can broken:))) cable or btt pi

Jbori347, 2024-06-16:
I've got the same problem. I'm able to view results of tests in the folders but have to restart klipper after every test

Frix-x, 2024-06-16:
I'm aware of this issue. It's not solved at the moment but shouldn't be very problematic as the graphs are done in background and should be generated correctly. So in the meantime, if the issue happens during the second axis (Y) measurement during the AXES_SHAPER_CALIBRATION macro, then just do only X, restart klipper and do Y to get both of the graphs generated.

I'm working on solving this. But the common point for people having this issue is running either a low powered pi clone like a BTT CB1 or a lot of things like CANbus, high precision stepper control, etc...

Frix-x, 2024-06-16:
I'll close this issue in favor of #120 that is the same

closed 2024-06-16 by Frix-x