Frix-x/klippain-shaketune #125: Using shaketune with a Beacon probe/acceleration sensor and defining axes_map

hapklaar, 2024-06-17:
### Describe the feature or hardware support you'd like

This might be more a question than a feature request:

After doing the AXES_MAP_CALIBRATION I get the detected axes map -x, -y, z. The doc states:
```
Use this value in your printer.cfg config file:

[adxl345] # replace "adxl345" by your correct accelerometer name
axes_map: -z,y,x
```
So tried adding it like this:

```
[beacon]
axes_map: -x,-y,z
```

But axes_map is not valid in the beacon section:
`Option 'axes_map' is not valid in section 'beacon'`

Probably doing something wrong. Also curious, does it matter if this has not been entered for other functionality of the script?

Frix-x, 2024-06-18:
As per [beacon documentation](https://docs.beacon3d.com/config/), it seems that it is:

```
[beacon]
accel_axes_map: x, y, z
#   Reassign or invert axes of the accelerometer to align with machine coordinates.
```

closed 2024-06-18 by Frix-x

hapklaar, 2024-06-18:
omg how did I miss that... thanks for pointing it out 👍