Frix-x/klippain-shaketune #126: Missing macros and cfg file?

vagudo, 2024-06-17:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

??

### Describe the bug and expected behavior

I am new to Shake'n'tune but I used it a couple of weeks ago and discovered my belts were unequal. I fixed this and got to troubleshooting some other things. After running some updates today I can't find the macros, and digging a bit more I can't even fine any .cfg regarding shake'n'tune...
I have tried running the install command on Putty but no luck.

What do I do?

Frix-x, 2024-06-18:
Please read the release notes from v4.0.0 : https://github.com/Frix-x/klippain-shaketune/releases
This version contain a breaking change due to the fact that it is now a Klipper plugin. Doing the required actions should fix your problem :)

closed 2024-06-18 by Frix-x

vagudo, 2024-06-18:
You are a rock star!!