Frix-x/klippain-shaketune #128: axes_map determination very inconsistent

hapklaar, 2024-06-18:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v4.0.1-0-g69ad2283

### Describe the bug and expected behavior

When trying to determine the axes map of my printer using the AXES_MAP_CALIBRATION command, I get varying or no results at all. The map was -x, -y, z yesterday. Today I've seen all kinds of variants and currently the map can't be determined correctly anymore. Printer (CoreXY RatRig400) seems to function just fine other than this.

Not sure if this is a problem with the printer, accelerometer or the script. I've seen the following axis maps:
-x,-y,z (I assume this is the correct one)
-x,-y,-z
x,y,z

Some varying results below:

```
12:13:11
--> Detected axes_map: unable to determine it correctly!
12:13:11
--> Detected gravity: 10.29 m/s²
12:13:11
Dynamic noise level: 285.01 mm/s² -> OK
12:13:11
Machine axis Z -> nearest accelerometer direction vector: [0 1 0] (angle error: 44.34°)
12:13:05
Machine axis Y -> nearest accelerometer direction vector: [0 1 0] (angle error: 4.60°)
12:13:03
Machine axis X -> nearest accelerometer direction vector: [1 0 0] (angle error: 1.66°)
12:13:00
This may take some time (1-3min)
12:13:00
Analysis of the movements...
12:12:52
Shake&Tune version: v4.0.1
12:12:51
AXES_MAP_CALIBRATION
```

```
12:26:18
--> Detected axes_map: -x, -y, z
12:26:18
--> Detected gravity: 10.29 m/s²
12:26:18
Dynamic noise level: 298.72 mm/s² -> OK
12:26:18
Machine axis Z -> nearest accelerometer direction vector: [0 0 1] (angle error: 37.12°)
12:26:12
Machine axis Y -> nearest accelerometer direction vector: [ 0 -1 0] (angle error: 5.32°)
12:26:10
Machine axis X -> nearest accelerometer direction vector: [-1 0 0] (angle error: 3.03°)
12:26:08
This may take some time (1-3min)
12:26:08
Analysis of the movements...
12:25:58
Shake&Tune version: v4.0.1
12:25:58
AXES_MAP_CALIBRATION
```

```
12:27:20
--> Detected axes_map: -x, -y, -z
12:27:20
--> Detected gravity: 10.30 m/s²
12:27:20
Dynamic noise level: 309.24 mm/s² -> OK
12:27:20
Machine axis Z -> nearest accelerometer direction vector: [ 0 0 -1] (angle error: 50.36°)
12:27:14
Machine axis Y -> nearest accelerometer direction vector: [ 0 -1 0] (angle error: 8.11°)
12:27:11
Machine axis X -> nearest accelerometer direction vector: [-1 0 0] (angle error: 4.65°)
12:27:09
This may take some time (1-3min)
12:27:09
Analysis of the movements...
12:26:59
Shake&Tune version: v4.0.1
12:26:59
AXES_MAP_CALIBRATION
```

### Additional information and klippy.log

[klippy.log](https://github.com/user-attachments/files/15884736/klippy.log)

Frix-x, 2024-06-18:
Can you also share your graphs please?

hapklaar, 2024-06-18:
Certainly. I use a beacon probe by the way, probe_points: 200,200,20

3 consecutive results:
[image]
[image]
[image]

roykrikke, 2024-07-07:
[image]
[image]
I have more or less the same issue. But I'm not sure if the spread on the Z axis has something to do with it.

Frix-x, 2024-07-12:
Hello,

I've done some diagnostic on this and I definitely need to improve the system. At the moment it only works in the case where the accelerometer is not moving that much (not vibrating/rotating). The problem is that most nozzle mounts or beacon kind of accelerometers will exhibit this kind of movements and are not perfect for this.

I have some ideas to change a bit the way I'm recording things in order to improve the measurement and analysis. I'll let you know when I've a testing branch available.

matthew-humphrey, 2024-07-17:
I'm having the same issue with the accelerometer built into an EBB36 v1.2. Fresh install. 
[image]

jeji, 2024-10-12:
hi there, I have same issue report. 
The HW is Voron Trident, SB2209 toolhead with adxl345
while look at the graphic, the first impression would be that Z axes confuse the resoult? As sensor is  actually not involved while Z axes moving
Anyway, thanks for the job, really good tool 

20:17
Cleaned up the output folder (only the last 3 results were kept)!
20:17
axes map graphs created successfully!
20:17
--> Detected axes_map: unable to determine it correctly!
20:17
--> Detected gravity: 9.46 m/s²
20:17
Dynamic noise level: 286.75 mm/s² -> OK
20:17
Machine axis Z -> nearest accelerometer direction vector: [0 1 0] (angle error: 23.23°)
20:17
Machine axis Y -> nearest accelerometer direction vector: [ 0 -1  0] (angle error: 4.68°)
20:17
Machine axis X -> nearest accelerometer direction vector: [0 0 1] (angle error: 3.95°)
20:17
This may take some time (1-3min)
20:17
Analysis of the movements...
20:16
Shake&Tune version: v4.1.0-1-g66f5e32
20:16
AXES_MAP_CALIBRATION

[image]

KD-4-Lyfe, 2024-12-27:
im not sure if this is related or not but i have just redone mine and installed the TMC_autotune and it worked first go

AJ6CU, 2025-01-19:
I had what appears to be the same problem. Like KD-4-Lyfe I installed and ran TMC_autotune and the axes were successfully mapped.

Mark

closed 2026-01-16 by Frix-x