Frix-x/klippain-shaketune #130: Out of bounds error. Error while generating the graphs: index 147 is out of bounds for axis 0 with size 125

kot0005, 2024-06-19:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

latest

### Describe the bug and expected behavior

fails to convert csv files to png
[klippy (6).log](https://github.com/user-attachments/files/15901399/klippy.6.log)
[image]
[image]

Also the belt comparison test varies by a lot run to run.

Frix-x, 2024-06-27:
Hello, I've pushed a fix for a potential bug in the CSV file writing function in v4.0.2 and your issue might be solved. Could you please update and let me know if it's still a problem?

Frix-x, 2024-06-30:
Hello, this was solved in v4.1.0

closed 2024-06-30 by Frix-x