Frix-x/klippain-shaketune #132: project.name contains an illegal character

Laikulo, 2024-06-24:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

N/A

### Describe the bug and expected behavior

The project.name specified in pyproject.toml contains an amperstand.

https://github.com/Frix-x/klippain-shaketune/blob/90ed7aca3c70b7a208ab65d3e1a9284f1f6be122/pyproject.toml#L2

Names must begin/end with an alphanumeric, and may contain dots, hyphens, and underscores in other positions (https://peps.python.org/pep-0508/)

```
identifier_end = letterOrDigit | (('-' | '_' | '.' )* letterOrDigit)
identifier    = letterOrDigit identifier_end*
name          = identifier
```

Laikulo, 2024-06-24:
(this breaks some other tooling in the python ecosystem, I found this initially on this fork (where I toss it unceremoniously in to a wheel), but it is also present on `main`)

https://github.com/Laikulo/shaketune/tree/packaged

Frix-x, 2024-06-27:
Thanks for raising it. I think it's not problematic in the current state of S&T since I don't really use this field currently. But I'll fix it to make it more clean :)

closed 2024-06-27 by Frix-x