Frix-x/klippain-shaketune #133: Huge variability in Belt tension results If Shake tune in run back to beak without a hard reboot.

kot0005, 2024-06-25:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

4.0.0-2-gfb8e1ce

### Describe the bug and expected behavior

I am getting very wild results in Belt tuning results if I run it back to back. Sometimes the Estimated similarity can swing to upto -10,000%

Rebooting before another run, seems to fix this issue.

[image]

[klippy (7).log](https://github.com/user-attachments/files/15970740/klippy.7.log)

[image]

[image]

[image]

Frix-x, 2024-06-25:
Can you share the full picture please (and the CSVs if you have them?)?

I would like to see how the graphs looks to understand why these numbers...

kot0005, 2024-06-26:
> Can you share the full picture please (and the CSVs if you have them?)?
> 
> I would like to see how the graphs looks to understand why these numbers...

Hi, where are the CSV files stored in the new version of Klippain shake tune ?

unfortunately i don't have all the graphs showing the -ve XXXXX% values. I found one which has a -115% similarity.

I only have the +ve XX% value graphs now. I have been rebooting Klipper every attempt, Which fixed the issue -ve % graphs in the 100's to 1000's

![image](https://github.com/Frix-x/klippain-shaketun
[image]
e/assets/68181325/db095e2f-9c6e-4f0c-83b8-389842ac1f19)
[image]

[image]

Frix-x, 2024-06-30:
Hello, thanks for sharing this. The issue should be solved in v4.1.0 as small bugfixes were done on the belt graph and how the similarity is computed :)

closed 2024-06-30 by Frix-x