Frix-x/klippain-shaketune #139: AXES_SHAPER_CALIBRATE gives different results than SHAPER_CALIBRATE

celtare21, 2024-07-02:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v4.1.0-1-g66f5e32

### Describe the bug and expected behavior

The results given by AXES_SHAPER_CALIBRATE from shaketune are quite different than the results given by klipper's SHAPER_CALIBRATE.
Don't know if it's also an issue, but the given dampening_ratio values are also quite different than the ones given by [ResHelper](https://github.com/lhndo/ResHelper).

SHAPER_CALIBRATE results:
[image]
[image]

AXES_SHAPER_CALIBRATE results:
[image]
[image]

Also the calculated dampening ratios from ResHelper are:
damping_ratio_x: 0.0085
damping_ratio_y: 0.0388

Frix-x, 2024-07-03:
Hi, thanks for reaching out and creating a topic about this. It is indeed something I need to add to my documentation: that's normal.

The explanation for the different filter suggestions is that the default Klipper scripts use forced values (0.1 and 5) for the damping ratio and square corner velocity (ResHelper does the same, AFAIK), even if they don't correspond to measured values on your machine.

But in Shake&Tune, since I retrieve the configured SCV and calculate the damping ratio from the graphs, I have the ability to inject them into the filter calculation algorithm. Using the real printer values has the effect of producing filter results that can be a bit different from the original scripts, especially if your real values are anything other than 0.1 and 5, as it's most of the time the case with Voron and other printers. The forced values are more suitable for Ender3, etc.. But the new computed filters done that way are also more representative of what really happens in your machine and S&T should end up being more accurate than the other tools.

Regarding the damping ratio, the number I give in S&T is an estimate. It's actually pretty hard to get a measurement for this based solely on a PSD graph, and that only works correctly if there is a main peak well separated from anything around it (and that's for example not the case in your X-axis graph), so the value is probably a bit eronned. But this is also true for ResHelper, and since it uses a different algorithm than I do, it's also giving you different values than in S&T (but it doesn't mean that one is better than the other, they're both estimates).
FYI, the algorithm I'm using for the damping ratio is based on a NASA paper and is called the Half-Quadratic Gain method. This method should be more accurate than the Half-Power method usually used by most tools, especially for higher damping ratio values (above 0.05) where we are most of the time.

celtare21, 2024-07-03:
Alright, thank you for the explanation!

closed 2024-07-03 by celtare21