Frix-x/klippain-shaketune #140: COMPARE_BELTS_RESPONSES Z_HEIGHT variable from variables.cfg not used

roykrikke, 2024-07-07:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

 Klippain  v4.3.1-17-gdf5ad5bf  Klippain-ShakeTune  v4.1.0-1-g66f5e32e

### Describe the bug and expected behavior

When I run COMPARE_BELTS_RESPONSES with the default parameters. I get the error: _Z_HEIGHT parameter is required if the test_point in [resonance_tester] section is set to -1,-1,-1_

When I run it with: _COMPARE_BELTS_RESPONSES FREQ_START= FREQ_END= HZ_PER_SEC=1 ACCEL_PER_HZ= TRAVEL_SPEED=120 Z_HEIGHT=50_
All works fine.

This is a bit confusing because there are parameters included in variables.cfg. I may be making a wrong assumption, but I expected these parameters to be used for this macro. Is this a mistake or am I making a wrong assumption?

```
## Resonnance testing position (if an accelerometer is installed in the machine)
## If test point is let to -1,-1, it default to the center of the bed at 50mm high
variable_resonnance_test_point_xy: -1, -1
variable_resonnance_test_z_clearance: 50
```
**Versions:**
Klippain: v4.3.1-17-gdf5ad5bf
Klippain-ShakeTune: v4.1.0-1-g66f5e32e

roykrikke, 2024-07-07:
The same is applicable to AXES_SHAPER_CALIBRATION

Frix-x, 2024-07-09:
Hello,

Yes you are right, I didn't updated the main Klippain repository yet in regards to the latest changes in Shake&Tune. I'll do it very soon

Benkult, 2024-09-27:
Y'all know you can put Z_Height parameter in the drop down list on the macro button, right ?

roykrikke, 2024-12-27:
@Frix-x  will you update this issue in the new 5.0 release also?