Frix-x/klippain-shaketune #141: unbound variable when running install.sh

eyenotion, 2024-07-09:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v4.1.0-c12653e

### Describe the bug and expected behavior

Getting an error when running the installation script, here's a copy of the console output;

=============================================
- Klippain Shake&Tune module install script -
=============================================

[PRE-CHECK] Klipper service found! Continuing...

libopenblas-dev is already installed
libatlas-base-dev is already installed
[DOWNLOAD] Downloading Klippain Shake&Tune module repository...
Cloning into 'klippain_shaketune'...
remote: Enumerating objects: 1461, done.
remote: Counting objects: 100% (818/818), done.
remote: Compressing objects: 100% (364/364), done.
remote: Total 1461 (delta 579), reused 572 (delta 451), pack-reused 643
Receiving objects: 100% (1461/1461), 38.43 MiB | 5.75 MiB/s, done.
Resolving deltas: 100% (903/903), done.
[DOWNLOAD] Download complete!

/home/pi/klippy-env/bin/activate: line 57: PS1: unbound variable

eyenotion, 2024-07-09:
Nevermind. I was using the script link on https://github.com/Frix-x/klippain-shaketune/blob/main/README.md. Used the one on https://github.com/Frix-x/klippain/blob/main/README.md and it installs fine.

Frix-x, 2024-07-12:
Hello,

This error happens when running in a Python 2 environment that is not supported by Shake&Tune. Python 2 is EOL since a couple of years now and I would suggest you to update your OS and/or Klipper environment to something more recent.

S&T needs Python >3.8 to work properly

eyenotion, 2024-07-13:
Thanks!

I was running python 3.7. Updated and working great now.

closed 2024-07-13 by eyenotion