Frix-x/klippain-shaketune #142: Error Installing collected packages: matplotlib, GitPython

AmshaSharam, 2024-07-09:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v4.1.0

### Describe the bug and expected behavior

i get this error on instalation:
Installing collected packages: matplotlib, GitPython
ERROR: Could not install packages due to an OSError: [Errno 13] Permission denied: '/home/pi/klippy-env/lib/python3.9/site-packages/mpl_toolkits/axes_grid1/__init__.py'
Check the permissions.

Frix-x, 2024-07-12:
Hello,

I think you are running RatOS. If that's the case, then it's a known bug on their side that was already raised to them.
See here https://github.com/Frix-x/klippain-shaketune/issues/116 for more information

AmshaSharam, 2024-07-15:
You are right, thank you.

closed 2024-07-16 by Frix-x