Frix-x/klippain-shaketune #143: Shaketune listed as a not valid config section after fresh install

stluck, 2024-07-13:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

NA

### Describe the bug and expected behavior

I've just installed Shake & Tune for the first time. Followed the ssh installation. After completing the installation I've added the [shaketune] section to my printer.cfg. On restart however, I'm getting this error: Section 'shaketune' is not a valid config section. 

When rerunning the ssh installation my output is as follows:

wget -O - https://raw.githubusercontent.com/Frix-x/klippain-shaketune/main/install.sh | bash
--2024-07-13 22:19:41--  https://raw.githubusercontent.com/Frix-x/klippain-shaketune/main/install.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 2606:50c0:8002::154, 2606:50c0:8001::154, 2606:50c0:8003::154, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|2606:50c0:8002::154|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 4863 (4.7K) [text/plain]
Saving to: ‘STDOUT’

-                   100%[===================>]   4.75K  --.-KB/s    in 0.001s  

2024-07-13 22:19:42 (5.93 MB/s) - written to stdout [4863/4863]

=============================================
- Klippain Shake&Tune module install script -
=============================================

[sudo] password for pi: 
[PRE-CHECK] Klipper service found! Continuing...

libopenblas-dev is already installed
libatlas-base-dev is already installed
[DOWNLOAD] Klippain Shake&Tune module repository already found locally. Continuing...

/home/pi/klippy-env/bin/activate: line 57: PS1: unbound variable

### Additional information and klippy.log

From my log:

Config error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 175, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 145, in _read_config
    pconfig.check_unused_options(config)
  File "/home/pi/klipper/klippy/configfile.py", line 304, in check_unused_options
    % (section,))
Error: Section 'shaketune' is not a valid config section
webhooks client 1966813632: New connection
webhooks client 1966813632: Client info {'program': 'Moonraker', 'version': 'v0.8.0-377-g4fe89d5'}

Frix-x, 2024-07-14:
Hello, unfortunately you have this error during the installation : `/home/pi/klippy-env/bin/activate: line 57: PS1: unbound variable`

This usually appears when running the install script in a Python2 environement: Shake&Tune needs at least Python 3.9 to work correctly.
So if you OS have this version available, you can update your klipper env to use Python3 (the easier is to use KIAUH). But if your OS is too old, you will need to update it to a more recent version: Debian Buster is too old, but Bullseye or Bookworm are both OK.

recrudesce, 2024-07-15:
I have this too.  Python 3.9.2, clean install.  Running bullseye, fully updated.

```
Config error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 271, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 214, in _read_config
    pconfig.check_unused_options(config, error_on_unused)
  File "/home/pi/klipper/klippy/configfile.py", line 456, in check_unused_options
    raise error(
configparser.Error: Section 'shaketune' is not a valid config section
```

I've tried installing multiple times to no avail.

freighttt, 2024-12-17:
> I have this too. Python 3.9.2, clean install. Running bullseye, fully updated.
> 
> ```
> Config error
> Traceback (most recent call last):
>   File "/home/pi/klipper/klippy/klippy.py", line 271, in _connect
>     self._read_config()
>   File "/home/pi/klipper/klippy/klippy.py", line 214, in _read_config
>     pconfig.check_unused_options(config, error_on_unused)
>   File "/home/pi/klipper/klippy/configfile.py", line 456, in check_unused_options
>     raise error(
> configparser.Error: Section 'shaketune' is not a valid config section
> ```
> 
> I've tried installing multiple times to no avail.
> 
> EDIT: So I tried this again, but this time I deleted the existing klippain-shaketune local git cloned directory. Seems to fix it.

how did you do this? im sort of an idiot at this stuff