Frix-x/klippain-shaketune #161: Support Passing Chips as a Parameter

falo2k, 2024-10-03:
### Describe the feature or hardware support you'd like

Would it be possible to add a CHIPS parameter as with TEST_RESONANCES to select the chip we want to use where more are defined?  When using something like Danger Klipper's optional mcu for nozzle probes, you can't have the optional mcu defined in the accel_chip attribute of resonance_tester (as it sometimes doesn't exist yet), but you can pass the defined chip to TEST_RESONANCES using CHIP=.  It would also be handy even for fixed mount sensors, when you want to take readings from different positions and compare.

yzheka, 2024-12-30:
You should not pass CHIPS parameter, it is not needed. Instead of this you have to use axis where each chip is fixed to. Anyway this is the question to the Klipper firmware but not the shaketune

falo2k, 2024-12-30:
I suggest you read again.  One use case is where you may have multiple chips reflecting the same axes (e.g. nozzle mounted, board mounted, etc.) and want to compare the results against each other while looking for possible factors that are causing certain results.

It is absolutely not a question to Klipper firmware - it is a request for the same flexibility offered in calling native Klipper macros to be mirrored in shaketune.

falo2k, 2024-12-30:
Except ... they are?

https://www.klipper3d.org/Measuring_Resonances.html#measuring-the-resonances

yzheka, 2024-12-30:
> Except ... they are?
> 
> https://www.klipper3d.org/Measuring_Resonances.html#measuring-the-resonances
ahh, wow. I did'n know that)

Frix-x, 2025-05-25:
This was added in the latest commit and soon to be merged (#219 )

The default behavior is still to read the `[resonance_tester]` config to try to find the best suited chip for each axis and measurements. But if for some reason an user want to override and manually specify a chip, you can do it now using ACCEL_CHIP as a macro parameter.

closed 2025-05-25 by Frix-x

falo2k, 2025-05-25:
Lovely.  Ta very much!