Frix-x/klippain-shaketune #163: Array to big

spotless20, 2024-10-07:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v4.1.0-1-g66f5e32e

### Describe the bug and expected behavior

Error while generating the graphs: array is too big; `arr.size * arr.dtype.itemsize` is larger than the maximum possible size.
None

Im sure there was a report in the past but hasn't helped to reinstall it, the "bug" appiered again. 

Shaper used are DKs bleeding-edge-v2 with the smooth shapers. 

Also calls the error when the startingpoint is higher so theoretically less frequencies have to be tested. 

Maybe some1 have an idea :)

[image]

### Additional information and klippy.log

[klippy-41.log](https://github.com/user-attachments/files/17278360/klippy-41.log)

Frix-x, 2024-10-18:
Hi,

Unfortunately I can't find the error in the klippy.log you provided. Can you clear your klippy.log, run the test (with the error) and send it immediately afterwards to capture this exact moment in it?

Thanks a lot!

spotless20, 2024-10-18:
Hey, all fine. 

Just the Test now in the Log. 

I would guess that you have somewhere declared a max amount of samples or so which are allowed to be in the graph, and with the additional shaper somehow it's too much or so. Last time I had this error (randomly appeared) I reinstalled shake tune and its gone, 
[klippy-42.log](https://github.com/user-attachments/files/17436025/klippy-42.log)

Frix-x, 2024-10-20:
Unfortunately there's still no error in the log... So I just don't understand what's happening.
I see that you are using DK, can you try on standard Klipper please?

jackjohnsonuk, 2024-11-07:
I've had exactly the same error, and again, nothing in the klippy logs.  I'm also on Danger Klipper (base, not bleeding edge as the other person), so it might be related to that, but I'm not sure why that would change a variable on arr size.
[klippy (53).log](https://github.com/user-attachments/files/17667965/klippy.53.log)

snipervi2000, 2025-02-18:
I've had the exact same error, running standard Klipper and I can't see anything in the logs. 

[klippy (5).log](https://github.com/user-attachments/files/18855582/klippy.5.log)

mikejsmtih1985, 2025-02-19:
Received this error today, 
```
Error while generating the graphs: array is too big; `arr.size * arr.dtype.itemsize` is larger than the maximum possible size.
None
```
I’m not seeing any errors in the log either. 

[logs-20250218-205159.zip](https://github.com/user-attachments/files/18857168/logs-20250218-205159.zip)

Blue-RR, 2025-02-19:
I've got the same error as well. Tried re-install no luck. Belt compare seems to be ok.

[klippy_shake_tune.log](https://github.com/user-attachments/files/18859024/klippy_shake_tune.log)

[image]

mattp87, 2025-02-19:
same here - // Error while generating the graphs: array is too big; `arr.size * arr.dtype.itemsize` is larger than the maximum possible size.
// None

Klippain-ShakeTune - v5.0.0-6-g216509e8

[klippy (6).log](https://github.com/user-attachments/files/18871391/klippy.6.log)

cuiviemen, 2025-02-19:
I also suffer from this issue on vanilla klipper. Running `axes_shaper_calibration` for results either XY axes I get:
```
20:25:15
Error while generating the graphs: array is too big; `arr.size * arr.dtype.itemsize` is larger than the maximum possible size.
None
20:25:15
Detected a square corner velocity of 5.0 and a damping ratio of 0.050. These values will be used to compute the input shaper filter recommendations
```
Klipper v0.12.0-418-g0114d72a6
Klippain-ShakeTune v5.0.0-6-g216509e8

snipervi2000, 2025-02-20:
Hi All, After a little googling I saw this [article ](https://stackoverflow.com/questions/68862470/gaussianprocessregressor-valueerror-array-is-too-big-arr-size-arr-dtype-ite) and noticed a similar issue was caused by the use of 32 bit software, I reinstalled a 64bit version of linux onto my Raspberry Pi and this has now worked.

N.B. the screenshot below shows where you can see if you're running a 64 or 32 bit OS

[image]

mikejsmith1985, 2025-02-20:
@snipervi2000 thanks for sharing this!

Telekatz, 2025-02-20:
I think the last commit caused the problem.
I had the same error message on my printer (32 bit OS). Downgrading the numpy and scipy libraries back to version 1.26.2 and 1.11.4 solved the problem.

Thmsdmsk, 2025-02-21:
> I think the last commit caused the problem. I had the same error message on my printer (32 bit OS). Downgrading the numpy and scipy libraries back to version 1.26.2 and 1.11.4 solved the problem.

Same

FordPrfkt, 2025-02-23:
> > I think the last commit caused the problem. I had the same error message on my printer (32 bit OS). Downgrading the numpy and scipy libraries back to version 1.26.2 and 1.11.4 solved the problem.
> 
> Same

This fixed it for me too, i am on a 32 bit system, so having a 64 bit kernel was not an option.

synfinatic, 2025-02-24:
> I think the last commit caused the problem. I had the same error message on my printer (32 bit OS). Downgrading the numpy and scipy libraries back to version 1.26.2 and 1.11.4 solved the problem.

fixed for me as well.

Frix-x, 2025-02-24:
Hello everyone, this is indeed a bug introduced by the latest commit.
After some investigation, it seems that Numpy 2.0 somehow broke 32 bit compatibility or at least updated some internal mechanisms about it. I'll see if I can fix my code to allow the update. In fact, a lot of packages are already updating, and numpy 2.0 is the version now used in Kalico, so I want to follow that trend to avoid having to downgrade the version in their venv and potentially breaking Klipper/Kalico in the future.

In the meantime, I'll revert the commit.

But I would strongly suggest using 64-bit OS on your hosts if available to be more future-proof, as I've seen 32-bit compatibility being dropped more and more by some Linux distributions, and every computer sold since around 2010 is 64-bit.

> This fixed it for me too, i am on a 32 bit system, so having a 64 bit kernel was not an option.

Just for my own curiosity and understand a bit more, can you tell me which kind of 32 bit host you are running ?

closed 2025-02-24 by Frix-x

synfinatic, 2025-02-24:
While I'm not on 32bit hardware (raspi 4B), I think part of the problem people run into is that MainsailOS recommends 32bit OS: https://docs-os.mainsail.xyz/getting-started/raspberry-pi-os-based

[image]

I know this is a common recommendation for RaspPi's since a 32bit OS supports up to 4GB of RAM and using a 64bit OS means using more memory.

Not saying you need to "fix" this, but it's probably worth clearly documenting.  Not sure if we can get MainsailOS/etc to update their docs or not.

FordPrfkt, 2025-03-02:
> Hello everyone, this is indeed a bug introduced by the latest commit. After some investigation, it seems that Numpy 2.0 somehow broke 32 bit compatibility or at least updated some internal mechanisms about it. I'll see if I can fix my code to allow the update. In fact, a lot of packages are already updating, and numpy 2.0 is the version now used in Kalico, so I want to follow that trend to avoid having to downgrade the version in their venv and potentially breaking Klipper/Kalico in the future.
> 
> In the meantime, I'll revert the commit.
> 
> But I would strongly suggest using 64-bit OS on your hosts if available to be more future-proof, as I've seen 32-bit compatibility being dropped more and more by some Linux distributions, and most computer sold since around 2010 is 64-bit. I say this because even when using a 64-bit host like a Pi 4 or equivalent, I have often seen people advise forcing the installation of the 32-bit flavor...
> 
> > This fixed it for me too, i am on a 32 bit system, so having a 64 bit kernel was not an option.
> 
> Just for my own curiosity and understand a bit more, can you tell me which kind of 32 bit host you are running ?

Sure, i am on a Orange Pi Zero LTS. I know a pretty low spec system but it runs my Voron Trident absolutely fine.

commit 9bb9937e (2025-02-24): reverted versionning changes to fix #163