Frix-x/klippain-shaketune #165: Error with multi klipper

pasha4ur, 2024-10-15:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

i don't know

### Describe the bug and expected behavior

Hello

If I have multiple instances of klipper then the installation script writes: "[ERROR] Klipper service not found, please install Klipper first!".

Each printer has different folder "PrinterName_data" and there is no "printer_data" folder

thhdragon, 2024-10-15:
have this also if you need more logs

Frix-x, 2024-10-18:
I'll close this issue as it's a duplicate of #12 
At the moment this is not supported as multi-instance Klipper is not an official way of installing Klipper.

But, I do plan to add this in the future and there's already some ongoing work. I'll keep you informed :)

closed 2024-10-18 by Frix-x