Frix-x/klippain-shaketune #166: install freeze at [Preparing metadata (pyproject.toml) ...]

SRAsolution, 2024-10-16:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v4.1.0

### Describe the bug and expected behavior

Unfortunately I'm unable to install [klippain-shaketune] since every time I run the install command the installation freeze a [Preparing metadata (pyproject.toml) ...

closed 2024-10-17 by SRAsolution

thhdragon, 2024-10-18:
what happened with this i had the same issue yesterday?

SRAsolution, 2024-10-18:
Hi,
Sorry I not a programmer and a good github user so I don't know how to
delete this topic

antecedent:
A few day ago I had to reinstall the OS in my raspberry (lite 64bit) I
updated everything than install klipper & try to install klippain...that
when the issue came.  I read on the internet that this issue could have
relations with python 3.12 and some people downgrade the version to bypass
the problem. I try my self but I think I made a bigger mess so I reinstall
the OS but now with the Mainsail preloaded version (python 3.9) and retry
to install klippain and it work.

SRAsolution, 2024-10-21:
> what was the solution? i'm having the same problem

Reinstall the raspberry with Mainsail OS (it had an older python version)
then directly install klipper measuring resonance and klippain.

The OS you will find under Operating system->Other specific-purpose OS->3d
printing->mainsail OS

therick0996, 2024-10-21:
deleting and reinstalling the entire OS sounds overkill. there has to be an easier way

SRAsolution, 2024-10-22:
Base on my previous research some people solve the issue by downgrading
they python version... I tried it but didn't work to me

ksteddom, 2024-10-22:
I had the same issue and found a fix. Initially I tried a fresh reinstall of the Raspberry Pi OS Lite 32 bit (not the Mainsail OS as mentioned above) but ran into the same issue. Specifically the install hangs indefinitely at the pip install of contourpy with the last line in the console being `Preparing metadata (pyproject.toml) ...`. Killing the process with ctrl z stops the install script but the system still runs very slow. I could not run the shutdown command to gracefully restart the system and ultimately had to unplug the pi to reboot.

**The problem:**

I am installing this on a Raspberry Pi 3 B with 1 GB of memory. When the install gets to contourpy it tries to install the latest version which was 1.3.0. It appears there is not a prebuilt whl for the combination of Python, libc, and architecture of the latest install so it tries to build a binary from source. Memory usage maxes out both the 1 GB of memory and the 1 GB of swap memory and the system thrashes. When ctrl z is used to stop the install, it stops the python script but does not kill the half dozen compiler tasks as seen in the top command.

**The solution:**
1. Reboot the pi. You should at least try ctrl z to kill the install and issue the shutdown command but you may find the system completely locked up as mine was and require unplugging the system.
2. Activate the venv that shaketune was using. `source ~/klippy-env/bin/activate`
3. Install an older version of contourpy. `pip install contourpy==1.2.1`
4. Restart your installation. When the installation script looks for requirements it will see that contourpy is already installed and use that version instead of the latest version.

This allowed me to complete the installation. I assume that the system will function correctly but haven't had time to test it fully. I'll report back in a few days.