Frix-x/klippain-shaketune #167: Erro on manual install

islonina, 2024-10-18:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v4.1.0-1-g66f5e32e

### Describe the bug and expected behavior

On manual install(change only direcotry for config folder and name of the moonraker and klipper-multi instance install) i on very end i got :
gcode command EXCITATE_AXIS_AT_FREQ already registered

islonina, 2024-10-18:
[klippy (2).log](https://github.com/user-attachments/files/17435357/klippy.2.log)

Frix-x, 2024-10-24:
Hello, multi-klipper instances are not officially supported at the moment... So it will be very difficult to help you debug this as the install script wasn't used.

Support is planned in the future.

closed 2024-11-13 by Frix-x