Frix-x/klippain-shaketune #169: Issue during install

luco85, 2024-10-23:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

not installed

### Describe the bug and expected behavior

Hi I'm trying to install script bui I have this error

[PRE-CHECK] Klipper service found! Continuing...

libopenblas-dev is already installed
libatlas-base-dev is already installed
[DOWNLOAD] Downloading Klippain Shake&Tune module repository...
Cloning into 'klippain_shaketune'...
remote: Enumerating objects: 1656, done.
remote: Counting objects: 100% (714/714), done.
remote: Compressing objects: 100% (298/298), done.
error: 7743 bytes of body are still expectedB | 160.00 KiB/s
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: index-pack failed
[ERROR] Download of Klippain Shake&Tune module git repository failed!

Voron 2.4 CB1 + MP8 V2
[klippy.log](https://github.com/user-attachments/files/17495305/klippy.log)

Frix-x, 2024-10-24:
Hello,

This error is usually due to an unstable wifi connection or a network problem. Please connect an Ethernet cable to your pi and try again.

closed 2024-10-24 by Frix-x

luco85, 2024-10-24:
Ok! I'll try. Thanks