Frix-x/klippain-shaketune #170: Measure relative accelerations using two IMU's.

gammeflamme, 2024-10-27:
### Describe the feature or hardware support you'd like

<h3>Relative acceleration</h3> 

I want to have a feature to measure the relative acceleration of the bed and toolhead during axes shaper calibration by attaching IMU's to both the bed and the toolhead. By subtracting the accelerations of one from the other we should be able to remove any potential resonances when they move in sync and show any resonances that are induced in the non active axis(the bed on corexy machines).

There are the two uses I could come up with for this.

<h3>Floppy Z axis </h3>
For example when bedslingers accelerate the bed it might cause the entire printer to move when this happens if the Z axis isn't completely rigid it will cause the Z axis to wobble causing another resonant frequency that can only be observed on the toolhead.

<h3>Floating/dampened printers</h3> 
  
Floating printers are in the ideal case printers that has full freedom to move in all directions. Because of this the both the bed and the nozzle needs to be measured as the bed is not stationary. More info about floating printers in additional context.
Finding a solution to this should not only make hanging printers viable it might also benefit printers that stand on wobbly surfaces or printers that use dampening feet such as hula.

<h3>My hanging printer</h3> 

[image]

        
<h3>Measurements </h3>
Here I have axis shaper calibration with IMU mounted to both the toolhead and the bed during both X and Y calibration

resonances were measured at 10mm Z
The axis map is based on IMU location not Axis measured

| Toolhead mounted IMU  | Bed mounted IMU |
| ------------- | ------------- |
| [image]  | [image]  |
| [image] | [image]  |

Some points of interest are

- The 3 Z peaks on the 2nd image that do not appear on the 1st. I assume that the 1st peak is the resonant frequency of the Y axis aluminum plate carriage as it appears in the Y test too but the other ones i don't know about, the 4th peak might be the ressonant frequency of the printer base frame.

- The 4th peak at the 3rd image which I suspect is my Z axis resonant frequency cause its a peak that only appears at the toolhead even though that test is moving the bed.

- That the 1st Z peak on both the 3rd and 4th image is at the same frequency and similar amplitudes so this might be a synced oscillation meaning that its irrelevant for inputshaper calibration.

As a last point I want to say that I might be completely wrong about all this and that I have misunderstood the entire thing.

### Additional context or information

Floating printers

The reason for this setup is that I have been experimenting with dampening feet on my printer and ended up just hanging the printer from above making whisper quiet and isolating it from my wobbly shelf. I have spent this last week trying to find good values for input shaping for my printer using an adxl345 but am yet to find any good values.
          
   Some potential pros and cons and of floating printers.
   <h4> Pros</h4> 
          
  - They don't resonate what they are standing on causing them to in most cases be quieter.
  - They don't inherit the resonant frequency of what they are standing on possibly improving performance.
  - The freedom might decrease the amount and/or amplitude of resonant frequency's of axis's.
  - The ability to hoist printers up and down to save space.
          
   <h4> Cons</h4> 
          
   - The vibrations the printer generates might have nowhere to go other than into moving the printer which might cause amplified vibrations affecting print quality.
   - The freedom might increase the amount and/or amplitude of resonant frequency's of axis's.
   - Difficulty of adequately either constraining or freeing the printer in the Z direction if this is required.
   - There might not be an implemented input shaper type that is applicable to them.
   - On bedslingers the Z axis stiffness in the Y direction becomes important because the base starts moving in the y direction

pylorak, 2024-11-20:
> it might also benefit printers that stand on wobbly surfaces or printers that use dampening feet such as hula.

Definitely, and these are the main reasons this feature would be very welcome. While the concept of hanging printers might be fun to experiment with, it is not something people typically use. But that is not the case with wobbly surfaces or dampeners. Those are very common.

It has been shown that the HULA (as an example), although effective in reducing vibrations and noise, often leads to a slight reduction in print quality. It is speculated that the reason is it changes the vibration profile in a way that is invisible to input shaper measurements, so the input shaper is always wrongly tuned. The reason is exactly as described by @gammeflamme: The input shaper must only correct the vibration of the toolhead **relative to the bed**, but accelerometers used for measurements measure absolute movements (compared to "stationary ground").

gammeflamme, 2024-11-21:
bwnanc over at the dangerklipper repo has experimented with using 3 IMU's to compensate over on the [3_point_shaper_branch](https://github.com/DangerKlippers/danger-klipper/tree/3_point_shaper_calibrate) branch of danger klipper by mounting a third IMU to the print frame and using that as a base line. 
I tried to modify this code to subtract the raw values from the toolhead and bed IMU's instead of using the resonance output but because the IMU's don't report the same amount of data points you either need to interpolate the data or force them to create the same amount of data points somehow. Which is why bwnanc used the computed resonance data instead but I think this muddels the data a fair bit. 
I managed to mount my printer reasonably stationary with a concrete slab but I still think the principle has a lot of potential, as we currently don't actually know what resonances the printer is actually exhibiting. 
If someone would like to collaborate on implementing this I would offer any assistance I can give. I have 3 IMU's and some basic programming and math skills. Both the interpolation and forcing the same amount of data points was outside my knowledge which is why I failed in my implementation.

pylorak, 2024-11-21:
> because the IMU's don't report the same amount of data points you either need to interpolate the data or force them to create the same amount of data points somehow.

Not only that, but when operating on the raw data you'd also have to make sure that the IMU's are reasonable synchronized in time.