Frix-x/klippain-shaketune #172: 'ResonanceTester' object has no attribute 'test'

TheMeanCanEHdian, 2024-12-07:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v4.1.0-1-g66f5e32e

### Describe the bug and expected behavior

When attempting to run Shake & Tune on the latest Klipper version I get the error `'ResonanceTester' object has no attribute 'test'`.

[image]

Could this be related to the recent Klipper behavior change for `test_resonances`?
https://github.com/Klipper3d/klipper/commit/16b4b6b302ac3ffcd55006cd76265aad4e26ecc8

### Additional information and klippy.log

[klippy.log](https://github.com/user-attachments/files/18049391/klippy.log)

mustang5269, 2024-12-08:
[klippy-2.log](https://github.com/user-attachments/files/18052647/klippy-2.log)

Same issue as well.

dclift97, 2024-12-08:
Also having the same issue `Internal Error on WebRequest: gcode/script
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/webhooks.py", line 256, in _process_request
    func(web_request)
  File "/home/pi/klipper/klippy/webhooks.py", line 436, in _handle_script
    self.gcode.run_script(web_request.get_str('script'))
  File "/home/pi/klipper/klippy/gcode.py", line 230, in run_script
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/pi/klipper/klippy/gcode.py", line 212, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/gcode.py", line 140, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/klipper/klippy/extras/gcode_macro.py", line 189, in cmd
    self.template.run_gcode_from_command(kwparams)
  File "/home/pi/klipper/klippy/extras/gcode_macro.py", line 68, in run_gcode_from_command
    self.gcode.run_script_from_command(self.render(context))
  File "/home/pi/klipper/klippy/gcode.py", line 227, in run_script_from_command
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/pi/klipper/klippy/gcode.py", line 212, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/gcode.py", line 140, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/klipper/klippy/extras/shaketune/shaketune.py", line 162, in cmd_COMPARE_BELTS_RESPONSES
    compare_belts_responses(gcmd, self._pconfig, st_process)
  File "/home/pi/klipper/klippy/extras/shaketune/commands/compare_belts_responses.py", line 26, in compare_belts_responses
    min_freq = gcmd.get_float('FREQ_START', default=res_tester.test.min_freq, minval=1)
                                                    ^^^^^^^^^^^^^^^
AttributeError: 'ResonanceTester' object has no attribute 'test'`

Masterpc96, 2024-12-08:
Also have the same issue :/

BOAndy1985, 2024-12-08:
This has to do with the new Klipper update, something was changed in the shaper calibration. I have the same problem.

Frix-x, 2024-12-08:
Yes this indeed a known problem. Klipper has merged a change in the resonance testing code (with a new kind of test) and Shake&Tune is impacted. So until I fix the compatibility, please either:
- Remove the `[shaketune]` section from your config and use the standard Klipper scripts
- Or keep Shake&Tune but don't update Klipper for now 

I'm at SMRRF so I'm not really able to fix it until I get back home at the end of the week unfortunately... But don't panic, I've identified the problem and will fix it as soon as possible 🙂

Venomx99, 2024-12-12:
From the Klipper Configuration Changes page.

20241203: The resonance test has been changed to include slow sweeping moves. This change requires that testing point(s) have some clearance in X/Y plane (+/- 30 mm from the test point should suffice when using the default settings). The new test should generally produce more accurate and reliable test results. However, if required, the previous test behavior can be restored by adding options sweeping_period: 0 and accel_per_hz: 75 to the [resonance_tester] config section.

Bomberski, 2024-12-15:
> From the Klipper Configuration Changes page.
>
> 20241203: The resonance test has been changed to include slow sweeping moves. This change requires that testing point(s) have some clearance in X/Y [...]

Did not work for me. Just wait for the fix

Tester196, 2024-12-15:
i have the same issue.

klipper
v0.12.0-401-g383b83d7

Klippain-ShakeTune
v4.1.0-1-g66f5e32e

[resonance_tester]
probe_points: 175, 175, 20
accel_chip: adxl345
sweeping_period: 0 
accel_per_hz: 75

[image]

Internal error on command:"AXES_SHAPER_CALIBRATION"
Internal Error on WebRequest: gcode/script
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/webhooks.py", line 256, in _process_request
    func(web_request)
  File "/home/pi/klipper/klippy/webhooks.py", line 436, in _handle_script
    self.gcode.run_script(web_request.get_str('script'))
  File "/home/pi/klipper/klippy/gcode.py", line 230, in run_script
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/pi/klipper/klippy/gcode.py", line 212, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/gcode.py", line 140, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/pi/klipper/klippy/extras/gcode_macro.py", line 189, in cmd
    self.template.run_gcode_from_command(kwparams)
  File "/home/pi/klipper/klippy/extras/gcode_macro.py", line 68, in run_gcode_from_command
    self.gcode.run_script_from_command(self.render(context))
  File "/home/pi/klipper/klippy/gcode.py", line 227, in run_script_from_command
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/pi/klipper/klippy/gcode.py", line 212, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/gcode.py", line 140, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/pi/klipper/klippy/extras/shaketune/shaketune.py", line 173, in cmd_AXES_SHAPER_CALIBRATION
    axes_shaper_calibration(gcmd, self._pconfig, st_process)
  File "/home/pi/klipper/klippy/extras/shaketune/commands/axes_shaper_calibration.py", line 26, in axes_shaper_calibration
    min_freq = gcmd.get_float('FREQ_START', default=res_tester.test.min_freq, minval=1)
AttributeError: 'ResonanceTester' object has no attribute 'test'

Frix-x, 2024-12-16:
I've started working on a fix. A first step is already available on https://github.com/Frix-x/klippain-shaketune/tree/fix-sweeping-compat

This first step is only to fix the error and allow the test to run correctly on new Klipper versions. However, it currently doesn't support the new sweeping test for now (I'm working on adding compatibility at the moment).

miklas206, 2024-12-20:
How do i install this fixed version?

eprouty, 2024-12-20:
```
cd ~/klippain_shaketune 
git fetch origin fix-sweeping-compat
git checkout fix-sweeping-compat
./install.sh
```

You may need to fix some config, looks like `keep_raw_csv` is now `keep_raw_data` in the `[shaketune]` section... but I think this will get you running on the fixed version @miklas206

miklas206, 2024-12-20:
> ```
> cd ~/klippain_shaketune
> git fetch origin fix-sweeping-compat
> git checkout fix-sweeping-compat
> ./install.sh
> ```
>
> You may need to fix some config, looks like `keep_raw_csv` is now `keep_raw_data` [...]

i did what you said but i still get the error tho. I have checked the cfg seems like it should be right so how do i know if it did update / replace the old one with the fixed version?

jimmycr, 2024-12-21:
> I've started working on a fix. A first step is already available on https://github.com/Frix-x/klippain-shaketune/tree/fix-sweeping-compat
>
> This first step is only to fix the error and allow the test [...]

confirm, that OpenNept4une (Elegoo N4+) with this quick fix works ;) Thanks!

dmit2k, 2024-12-21:
> This first step is only to fix the error and allow the test to run correctly on new Klipper versions. However, it currently doesn't support the new sweeping test for now (I'm working on adding compatibility at the moment).

fix works for my customised TwoTrees SK1 just fine, thank you!

Alia5, 2024-12-21:
> fix-sweeping-compat

Maybe you have cloned shaketune to "~/klippain-shaketune" (dash) instead of "klippain_shakletune" (underscore)

The Install-script doesn't exactly use the repo name and this might lead to issues........................................................

Alia5, 2024-12-21:
> i did what you said but i still get the error tho. I have checked the cfg seems like it should be right so how do i know if it did update / replace the old one with the fixed version?

Maybe you have cloned shaketune to "~/klippain-shaketune" (dash) instead of "klippain_shakletune" (underscore)

The Install-script doesn't exactly use the repo name and this might lead to issues........................................................

miklas206, 2024-12-22:
> Maybe you have cloned shaketune to "~/klippain-shaketune" (dash) instead of "klippain_shakletune" (underscore)
>
> The Install-script doesn't exactly use the repo name and this might lead to [...]

Im very sure about that it did it correctly i have tried to crosscheck the files and everything should be the fixed version but im still getting erros

Frix-x, 2024-12-22:
I'm adding the "done" label since everything is now functional with my latest commits on this branch. With the new code:
- Older Klipper version works as usual and only the "pulse" test is available
- Newer Klipper version allow the user to choose between sweeping or pulse-only test by changing the parameters in the `[resonance_tester]` config section (basically setting `sweeping_period: 0` to do the pulse-only test)
- The corresponding documentation was added to help you decide which test to use

This will be merged now into the develop branch to be released tonight :)

roykrikke, 2024-12-23:
Thank @Frix-x great work to get this fixed.

How do I get the develop branch on my system?
I running v4.1.0-1-g66f5e32e now with Klipper v0.12.0-404-g80d185c94-dirty.

Printfail, 2024-12-23:
How do you get the new 5.0 version ?

drumstick77, 2024-12-24:
Hi, has the latest v5.0 on develop branch been merged with main yet? I don't think it has, but I thought I checked.  thanks

roykrikke, 2024-12-24:
> Hi, has the latest v5.0 on develop branch been merged with main yet? Ii doesn't look like it has, but I thought I checked. thanks

That's why @Frix-x wrote it's still in develop branch.

drumstick77, 2024-12-24:
> > Hi, has the latest v5.0 on develop branch been merged with main yet? Ii doesn't look like it has, but I thought I checked. thanks
> 
> That's why @Frix-x wrote it's still in develop branch.

yeah, I thought so too, but the label done confuses me slightly.
Have you found the way to change moonraker to develop branch? thanks

roykrikke, 2024-12-24:
> yeah, I thought so too, but the label done confuses me slightly. Have you found the way to change moonraker to develop branch? thanks

I think you can just change the branch to develop and no change in moonraker.

https://www.freecodecamp.org/news/git-switch-branch/

PS I have not tried it yet :-)

miklas206, 2024-12-24:
> yeah, I thought so too, but the label done confuses me slightly. Have you found the way to change moonraker to develop branch? thanks

i made it work asking chatgpt about how i could change it to another branch and changed the moonraker config to the develop branch

roykrikke, 2024-12-24:
> i made it work asking chatgpt about how i could change it to another branch and changed the moonraker config to the develop branch

Please share the details. So all can learn :-)

HeavilyModdedEnder3, 2024-12-24:
> How do you get the new 5.0 version ?
This worked for me to go onto the pull request for v5.0.0
`cd ~/klippain_shaketune
git fetch origin pull/181/head:pr-181
git checkout pr-181
./install.sh`

lucidhack, 2024-12-25:
> > How do you get the new 5.0 version?
> 
> This worked for me to go onto the pull request for v5.0.0 `cd ~/klippain_shaketune` `git fetch origin pull/181/head:pr-181` `git checkout pr-181` `./install.sh`

Thanks!  This also worked for me, but I also needed to change:

This:
```
[shaketune]
keep_raw_csv = False
```

To This:
```
[shaketune]
keep_raw_data = False 
```

lucidhack, 2024-12-25:
> > How do you get the new 5.0 version?
> 
> This worked for me to go onto the pull request for v5.0.0 `cd ~/klippain_shaketune` `git fetch origin pull/181/head:pr-181` `git checkout pr-181` `./install.sh`

> > How do you get the new 5.0 version?
> 
> This worked for me to go onto the pull request for v5.0.0 `cd ~/klippain_shaketune` `git fetch origin pull/181/head:pr-181` `git checkout pr-181` `./install.sh`

Thanks!  This also worked for me, but I also needed to change:

This:
```
[shaketune]
keep_raw_csv = False
```

To This:
```
[shaketune]
keep_raw_data = False 
```

drumstick77, 2024-12-25:
Cheers all. I did run the install.sh previously, what I was missing was what @lucidhack mentioned about the change on [shaketune] under printer.cfg. it's now working.

one last question however, once it's been merged to main, I presume it will just appear in moonraker as per normal and running update from it should update the current state of shaketune?

clstr, 2024-12-26:
> Cheers all. I did run the install.sh previously, what I was missing was what @lucidhack mentioned about the change on [shaketune] under printer.cfg. it's now working.
>
> one last question however, [...]

You would do the same steps but the branch would be `main`.

DVSVIDEO, 2024-12-26:
Can someone explain to me why there are branches and if there is a working V5 why the old version is the last one up?
Why does it not update when I press search for updates? Shaketune is in the list.

drumstick77, 2024-12-26:
> You would do the same steps but the branch would be `main`.

Yeah, I get that. But we ran install.sh which I believe installed dev version without changing Moonraker cfg file.

Raouad, 2024-12-27:
> Don't know what I am doing. Tried
> 
> ```
> cd ~/klippain_shaketune
> git fetch origin pull/181/head:pr-181
> git checkout pr-181
> ./install.sh
> ```
> 
> and failed.
> 
> Will wait until the develop branch gets pulled into main. Any idea when this might happen?.

git clone https://github.com/Frix-x/klippain-shaketune.git
cd klippain-shaketune
git checkout develop
bash install.sh

Try this

patofoto, 2024-12-27:
@Raouad Thank you. I think I will wait. I don't want to be using patched solutions just to try now. I am in no real need to use this now. I was just curious. Can wait.

closed 2024-12-30 by Frix-x

kot0005, 2025-01-17:
why is this closed ? I just had this issue occur. What is the solution ? Thank you

roykrikke, 2025-01-17:
> why is this closed ? I just had this issue occur. What is the solution ? Thank you

Update to last version 5.x