Frix-x/klippain-shaketune #175: Internal error on command:"_AXES_SHAPER_CALIBRATION"

Printfail, 2024-12-16:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v4.1.0-1-g66f5e32e

### Describe the bug and expected behavior

Nach Starten des Marcos AXES_SHAPER_CALIBRATION geht der Drucker sofort in shutdown: Command request.
Hatte Klippian-Shaketune neu installieren müssen weil ich meinen Drucker neu aufsetzen musste. 

### Additional information and klippy.log

[klippy.log](https://github.com/user-attachments/files/18150686/klippy.log)

strayr, 2024-12-16:
This is also causing 
`Internal error on command:"_COMPARE_BELTS_RESPONSES"`
`Internal error on command:"_AXES_SHAPER_CALIBRATION"`

I think this is the klipper update 20241203
https://www.klipper3d.org/Config_Changes.html
It's this commit https://github.com/Klipper3d/klipper/commit/16b4b6b302ac3ffcd55006cd76265aad4e26ecc8

Simplest immediate solution for @Printfail and any othr end user is to roll back klipper to this commit https://github.com/Klipper3d/klipper/commit/7f89668d6c069deaa02d6479e073d11a2389a948 until there is an updated shaketune. 

**This breaks all of shaketune and is quite serious.**

Sotiocsi, 2024-12-16:
In klipper, they have changed this resonance measurement.
The print head moves during resonance measurement.

We are waiting for the update to shaketune.

Thank you.

Frix-x, 2024-12-16:
Hello,

This is a duplicate of #172 . I'll close this one in favor of the other.

The issue is known and identified, but I was on holiday until today and couldn't work on a fix until now. A new release will come very soon to fix it.

closed 2024-12-16 by Frix-x

Killajoedotcom, 2025-02-26:
I'm still getting this error on the newest 5.1.0. Any other ideas I could try?

Frix-x, 2025-02-26:
> I'm still getting this error on the newest 5.1.0. Any other ideas I could try?

I'm pretty sure it's not the same error as this one was already fixed for some time. Can you share your klippy.log and open a new issue with the exact error inside?

Thank you