Frix-x/klippain-shaketune #183: Graphs peak difference

DVSVIDEO, 2024-12-30:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v5.0.0-0-g74364d72

### Describe the bug and expected behavior

 "Axis Frequency Profile" and below
"Time -Frequency Histogram"  frequency peaks not equal
[image]

Frix-x, 2024-12-31:
Thank you, I may have introduced this inadvertently. I'll fix it in the next bugfixes release :)

Frix-x, 2025-02-24:
This is now fixed in the latest develop branch commit, ready for the next release :)

Frix-x, 2025-02-24:
Closing it now as it's released in v5.1.0