Frix-x/klippain-shaketune #184: ERROR  after update Option 'keep_raw_csv' is not valid in section 'shaketune'

Byourei, 2024-12-30:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v5.0.0-0-g74364d72

### Describe the bug and expected behavior

Option 'keep_raw_csv' is not valid in section 'shaketune'

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted

what it this?

thiagogssantos, 2024-12-31:
I had the same problem,
check out the "readme" , some settings need to be changed in printer.cfg

DVSVIDEO, 2024-12-31:
You didn't read that you need to copy new settings into printer.cfg instead of the old ones.

closed 2024-12-31 by Frix-x