Frix-x/klippain-shaketune #185: Decimal places for excite axis at frequency please

oernster, 2024-12-31:
### Describe the feature or hardware support you'd like

Currently only integers are supported in Excite Axis at freq.
However, Axes shaper calibrate supports floats for the peaks.
Please can you support floats for excite axes at freq as well?

Frix-x, 2025-01-01:
Thanks for the feature request. It's definitely doable, although I'm not sure of the use case. Usually a resonance peak is a peak that is a bit wide by a few hertz, not just at a specific decimal point.

But anyway, I can change the input field to accept floats in the next version for ease of use.

Frix-x, 2025-02-24:
Closing it now as it's released in v5.1.0

closed 2025-02-24 by Frix-x