Frix-x/klippain-shaketune #192: Shake&Tune was unable to retrieve accelerometer data in time

Xtrem69, 2025-01-02:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v 5.00

### Describe the bug and expected behavior

la version 4 aucun problèmes mais depuis la version 5 pour créer profil de vibration il me fait une erreur directement après le 2mm sec Shake&Tune was unable to retrieve accelerometer data in time. This might be due to slow hardware or a busy system. merci par avance

version 4 no problems but since version 5 to create vibration profile it gives me an error directly after 2mm sec Shake&Tune was unable to retrieve accelerometer data in time. This might be due to slow hardware or a busy system

Xtrem69, 2025-01-08:
je me sert de la sonde beacon pour mes test

Frix-x, 2025-02-24:
Hello, la partie qui s'occupe de ça a été complètement réécrite dans la dernière version sur develop. Donc il y a de grande chances pour que ton erreur soit résolue :)

Frix-x, 2025-02-24:
Closing it now as it's released in v5.1.0

closed 2025-02-24 by Frix-x

Xtrem69, 2025-02-24:
salut j 'espere car j etais obliger pour la parti vibration de prendre la version 4 ^^

Xtrem69, 2025-02-25:
salut toujours le  meme probleme

i3docdev, 2025-03-02:
Still having issues with this using scanner (cartographer) as the ADXL.

[message.txt](https://github.com/user-attachments/files/19044260/message.txt)

Xtrem69, 2025-03-02:
Beacon moi aussi par contre si je modifie pas la taille de bed ça passe étrangement

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>