Frix-x/klippain-shaketune #194: X/Y Axis aren't centered after A belt resonance measurement.

irebane, 2025-01-07:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

5.0.0

### Describe the bug and expected behavior

Running tests on Voron Trident 250mm

While running **Relative Belts calibration Tool** after A belt data measured and tool proceed with B Belt measurements toolhead is not swinging thru the same center point and it starts B Belt measurements from the position at which A belt measurements finished which is shifted both back and right ~10-15mm. It seems that eventually it leads to end result being shifted on the graph even belts have equal tension.

Thank you.

MBOKjr, 2025-01-13:
This is also present on the AXES_SHAPER_CALIBRATION. Same scenario, after first axis test, the toolhead is no longer centered during the next test.

Frix-x, 2025-01-13:
Yes, this is a known bug and my fault. I'll fix it on the next release

PR #196 (2025-02-09): recenter toolhead before each measurement
fixes #194

## Summary by Sourcery

Bug Fixes:
- Fix an issue where the toolhead was not centered before each measurement, resulting in inaccurate measurements.
closed 2025-02-09 by Frix-x