Frix-x/klippain-shaketune #195: [resonance_tester]

bombsquad85, 2025-01-07:
### Describe the feature or hardware support you'd like

Is it possible to not require [resonance_tester] in the printer.cfg?

reason being using the KUSBA or other nozzle mounted boards. the KUSBA has a separate adxl.cfg file that  gets included into the printer.cfg but when not in use or unplugged gets commented out. with the [resonance_tester] section required in the printer.cfg if the KUSBA is disconnected then you get an MCU error. 

just something to think about. 

Thanks!

bombsquad85, 2025-01-13:
Am I the only one experiencing this? 

Lol

ils15, 2025-01-27:
same error here

Frix-x, 2025-02-09:
Hi, [resonance_tester] is a required config section to run Shake&Tune as it reads the accelerometer config from here. So if you run Shake&Tune without it, it's normal that it crashes... So what you experience is a normal behavior.

If I remove this check, you will be able to remove the KUSBA cfg file and leave S&T enabled, but then you will have a Klipper panic when you click on the macros. So my advice here would be to put the S&T config directly into the KUSBA file, because there's no point in keeping S&T enabled if you remove the accelerometer, so better to disable it all at once.

closed 2025-02-09 by Frix-x