Frix-x/klippain-shaketune #198: Internal error during connect: module 'dateutil' has no attribute '__version__'

chalb0, 2025-01-27:
### K-Shake&Tune module branch

- [x] I confirm using the main branch

### Version

5.0

### Describe the bug and expected behavior

On install and use of [shaketune] in printer.cfg, klipper state becomes disconnected with the error:

> Klipper reports: ERROR
> 
> Internal error during connect: module 'dateutil' has no attribute '__version__'
> 
> Once the underlying issue is corrected, use the "RESTART"
> command to reload the config and restart the host software.
> Printer is halted

However dateutil is installed with an atribute in version:
```
$ python3 -c "import dateutil; print(dateutil.__version__)"
2.8.2
```

I have attempted a reinstall. 

I did notice that looking at the head of klippy.py it was showing python2, changing this for 3 still has the same issue.
```
$ head -n 1 /home/pi/klipper/klippy/klippy.py
#!/usr/bin/env python3
```

Any suggestions?

Frix-x, 2025-02-09:
Hello,

I'm pretty sure this is not Shake&Tune related since I don't use the dateutil library... Can you share a klippy.log with the error to validate its origin?

chalb0, 2025-02-13:
Hi, I've attached klippy log - I'm only getting the error when I unhash [shaketune]

[klippy.log](https://github.com/user-attachments/files/18782144/klippy.log)

Relevant part of the log:
```
=======================
Unhandled exception during connect
Traceback (most recent call last):
  File "/home/chalbo/klipper/klippy/klippy.py", line 130, in _connect
    self._read_config()
  File "/home/chalbo/klipper/klippy/klippy.py", line 123, in _read_config
    self.load_object(config, section_config.get_name(), None)
  File "/home/chalbo/klipper/klippy/klippy.py", line 103, in load_object
    mod = importlib.import_module('extras.' + module_name)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/chalbo/klipper/klippy/extras/shaketune/__init__.py", line 15, in <module>
    from .shaketune import ShakeTune as ShakeTune
  File "/home/chalbo/klipper/klippy/extras/shaketune/shaketune.py", line 23, in <module>
    from .graph_creators import GraphCreatorFactory
  File "/home/chalbo/klipper/klippy/extras/shaketune/graph_creators/__init__.py", line 22, in <module>
    from .axes_map_graph_creator import AxesMapGraphCreator as AxesMapGraphCreator  # noqa: E402
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/chalbo/klipper/klippy/extras/shaketune/graph_creators/axes_map_graph_creator.py", line 19, in <module>
    from .graph_creator import GraphCreator
  File "/home/chalbo/klipper/klippy/extras/shaketune/graph_creators/graph_creator.py", line 18, in <module>
    from matplotlib.figure import Figure
  File "/home/chalbo/klippy-env/lib/python3.11/site-packages/matplotlib/__init__.py", line 259, in <module>
    _check_versions()
  File "/home/chalbo/klippy-env/lib/python3.11/site-packages/matplotlib/__init__.py", line 254, in _check_versions
    if parse_version(module.__version__) < parse_version(minver):
                     ^^^^^^^^^^^^^^^^^^
AttributeError: module 'dateutil' has no attribute '__version__'
webhooks client 546864561808: New connection
webhooks client 546864561808: Client info {'program': 'Moonraker', 'version': 'v0.9.3-48-g354cc11'}
```

Networx666, 2025-02-15:
Hey I have the same Problem...

[klippy.log](https://github.com/user-attachments/files/18810860/klippy.log)

```
Unhandled exception during connect
Traceback (most recent call last):
  File "/home/marcel/klipper/klippy/klippy.py", line 130, in _connect
    self._read_config()
  File "/home/marcel/klipper/klippy/klippy.py", line 123, in _read_config
    self.load_object(config, section_config.get_name(), None)
  File "/home/marcel/klipper/klippy/klippy.py", line 103, in load_object
    mod = importlib.import_module('extras.' + module_name)
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/marcel/klipper/klippy/extras/shaketune/__init__.py", line 15, in <module>
    from .shaketune import ShakeTune as ShakeTune
  File "/home/marcel/klipper/klippy/extras/shaketune/shaketune.py", line 23, in <module>
    from .graph_creators import GraphCreatorFactory
  File "/home/marcel/klipper/klippy/extras/shaketune/graph_creators/__init__.py", line 22, in <module>
    from .axes_map_graph_creator import AxesMapGraphCreator as AxesMapGraphCreator  # noqa: E402
  File "/home/marcel/klipper/klippy/extras/shaketune/graph_creators/axes_map_graph_creator.py", line 19, in <module>
    from .graph_creator import GraphCreator
  File "/home/marcel/klipper/klippy/extras/shaketune/graph_creators/graph_creator.py", line 18, in <module>
    from matplotlib.figure import Figure
  File "/home/marcel/klippy-env/lib/python3.9/site-packages/matplotlib/__init__.py", line 259, in <module>
    _check_versions()
  File "/home/marcel/klippy-env/lib/python3.9/site-packages/matplotlib/__init__.py", line 254, in _check_versions
    if parse_version(module.__version__) < parse_version(minver):
AttributeError: module 'dateutil' has no attribute '__version__'
webhooks client 3980624080: New connection
webhooks client 3980624080: Client info {'program': 'Moonraker', 'version': 'v0.9.3-48-g354cc11'}
```

Frix-x, 2025-05-25:
Hello,

Was this issue solved ?
I think it's due to an incorrect matplotlib version or something like this.

chalb0, 2025-05-25:
Hi @Frix-x , thanks for checking back. 

I had another look and think I found the problem and a fix. The error has gone away but I've not had opportunity to test Shaketune yet.

You were right in that the issue was with **matplotlib** - although this I believe is/was bundled somewhere.

### What seems to be the fix:

1) Activate Python Environment
`source ~/klippy-env/bin/activate`

2) Reinstall broken package:
`pip uninstall python-dateutil -y`
`pip install python-dateutil`

3) Restart Klipper & check it works with:
`python -c "import dateutil; print(dateutil.__version__)"`