Frix-x/klippain-shaketune #201: Multi-Chip based Shake-Tune Capability

saeflizard, 2025-02-08:
### Describe the feature or hardware support you'd like

I recently assembled an LDO Voron Switchwire and would like to use 2 chips to measure resonances. I plan on using the LDO toolhead board to measure the X axis while using a BTT LIS2DW for the Y. Note that I am running Kalico for the ability to print with non-critical mcu connections. Is this something that would be able to be implemented?

Frix-x, 2025-02-09:
This should already be working correctly : just use accel_chip_x/y in the [resonance_tester] Klipper/Kalico config section:

```yaml
[resonance_tester]
#accel_chip:
#   A name of the accelerometer chip to use for measurements. If
#   adxl345 chip was defined without an explicit name, this parameter
#   can simply reference it as "accel_chip: adxl345", otherwise an
#   explicit name must be supplied as well, e.g. "accel_chip: adxl345
#   my_chip_name". Either this, or the next two parameters must be
#   set.
accel_chip_x:
accel_chip_y:
#   Names of the accelerometer chips to use for measurements for each
#   of the axis. Can be useful, for instance, on bed slinger printer,
#   if two separate accelerometers are mounted on the bed (for Y axis)
#   and on the toolhead (for X axis). These parameters have the same
#   format as 'accel_chip' parameter. Only 'accel_chip' or these two
#   parameters must be provided.
```

saeflizard, 2025-02-10:
Thanks for the assistance

closed 2025-02-10 by saeflizard