Frix-x/klippain-shaketune #208: add ACCEL_CHIP parameter

dderg, 2025-03-30:
### Describe the feature or hardware support you'd like

Printers wit a toolchanger setup need to run accelerometer measurement on every tool. When using a toolhead board accelerometer, you have to change the chip setting in the config and restart the firmware between each measurement. [motors-sync](https://github.com/MRX8024/motors-sync/tree/main) has a parameter `ACCEL_CHIP` which can be used to make a macro that uses a correct accelerometer for the current toolhead board. Would something like that be possible?

brian000, 2025-05-16:
This would be appreciated for Beacon users as well.

Frix-x, 2025-05-25:
This was done in the latest commit, soon to be merged and released (#219 )

closed 2025-05-25 by Frix-x