Frix-x/klippain-shaketune #209: Documentation for CLI

mdburkey, 2025-03-30:
### Describe the feature or hardware support you'd like

I saw that in version 5.0 and afterwards you said a proper command line interface for regenerating PNG's from the original data files had been added. Has any documentation been created on how to actually use the CLI, as I have serious need to be able to analyze CSV's generated on one printer (that isn't able to to install all the updates needed for ShakeTune) on a different newer Debian install that is able to.

Frix-x, 2025-05-25:
Hello,

Sorry for the delay, but yes the CLI was quite experimental. I was using it for development purpose but cleaned it in order to expose it for more public use in v5. No real documentation was made yet but the CLI was already pretty self explanatory by using the --help command.

But this was from the past as I just added a proper documentation to help you use it. It will be pushed on the next version :)

Frix-x, 2025-05-25:
See #219 

I'll close this now and the documentation will be merged soon.

closed 2025-05-25 by Frix-x