Frix-x/klippain-shaketune #210: input shaper file size causing unreadable results

michaelstarkie, 2025-03-31:
### K-Shake&Tune module branch

- [x] I confirm using the main branch

### Version

Klippain-ShakeTune v5.1.1-0-gd59da30a

### Describe the bug and expected behavior

hi just downloaded shake and tune last week for my vzbot 330 awd with octopus max ez and beacon. i can complete the belt analysis no problem but when i try the input shaker analysis i get a fault due to the size of the results (6.2mb) and the screen when i download it is obviously unreadable 

### Additional information and klippy.log

[image]

[image]

[image]

dontblink, 2025-04-30:
was having the same issue, i ran ./install in the klippain_shaketune and that downgraded my numpy, scipy, and matplotlib libs, which seemed to fix the issue for me
`Installing collected packages: smmap, numpy, scipy, PyWavelets, gitdb, matplotlib, GitPython
  Attempting uninstall: numpy
    Found existing installation: numpy 2.0.2
    Uninstalling numpy-2.0.2:
      Successfully uninstalled numpy-2.0.2
  Attempting uninstall: scipy
    Found existing installation: scipy 1.13.1
    Uninstalling scipy-1.13.1:
      Successfully uninstalled scipy-1.13.1
  Attempting uninstall: matplotlib
    Found existing installation: matplotlib 3.9.4
    Uninstalling matplotlib-3.9.4:
      Successfully uninstalled matplotlib-3.9.4
Successfully installed GitPython-3.1.41 PyWavelets-1.6.0 gitdb-4.0.12 matplotlib-3.8.2 numpy-1.26.2 scipy-1.11.4 smmap-5.0.2

[INSTALL] Klippain Shake&Tune Klipper module is already installed. Continuing...`

michaelstarkie, 2025-05-01:
Hindonyounuave a screenshot of where you ran. /install as still new to klipper and the programming side of things but certainly a big thanks for the advice

dontblink, 2025-05-01:
@michaelstarkie [image]