Frix-x/klippain-shaketune #214: Klipper disconnect at the end of belt test

LazaroFilm, 2025-04-16:
### K-Shake&Tune module branch

- [x] I confirm using the main branch

### Version

v5.1.1-0-gd59da30a

### Describe the bug and expected behavior

Shaketune (standalone) crashes at the end of the belt test. 
It also left a file calle `snt_tmp-f2b6004a.stdata` in the belt folder.

### Additional information and klippy.log

[klippy-3.log](https://github.com/user-attachments/files/19779580/klippy-3.log)

[image]

LazaroFilm, 2025-04-19:
This keeps happening with this one test. I've had successfully prints and the other tests are working properly but not the belts.

LazaroFilm, 2025-04-26:
I narrowed it down to it crashing when I disable the sweep motion.

staticanime, 2025-05-08:
Just to add, I have had similar. Whats happening on my system is the CAN toolhead board is disconnecting due to the main CPU being busy starting processing the csvs into graphs. I think a configurable delay of 5-15 seconds before starting "Belts comparative frequency profile generation" would avoid this?

LazaroFilm, 2025-05-16:
That might be it. I have a CB2 board which isn't the fastest in the lot.

Frix-x, 2025-05-25:
I can add a longer delay if needed. However, the current code already waits for the first belt's measurement to be saved and the file to be freed before proceeding to the second belt, so I'm not sure what the problem is.

This issue could be related to #124, which was resolved for most people. However, the garbage collection in Python/Klipper can still be erratic and cause problems with Klipper timings. Maybe that's the problem here! I'll look into adding a longer dwell time between the two belt measurements to see if that solves the problem for you!

Frix-x, 2025-05-25:
I added a 5 seconds dwell time between the two measurements on the belts macro. I think it's ok to add this for everyone and avoid complexifying the config. Let's see if it solve your issue !

staticanime, 2025-05-28:
I can report that the 5s dwell fixed my crashes!

LazaroFilm, 2025-10-06:
Just adding, after this issue I had some crashes, but they were from having different versions of Klipper flashed on the MCUs (my Night Owl had a more recent version than my main MCU). after reflashing both the issue went away.