Frix-x/klippain-shaketune #218: Install fails with unbound variable error

Trailglider, 2025-05-22:
### K-Shake&Tune module branch

- [x] I confirm using the main branch

### Version

N/A (install problem)

### Describe the bug and expected behavior

Trying to install on my Voron 2.4, Raspberry Pi 3B, mainsail and fully updated klipper version (v0.13.0-114-g841a9ca2)

When running the install it fails and I get the below output:
```
pi@mainsailos:~ $ wget -O - https://raw.githubusercontent.com/Frix-x/klippain-shaketune/main/install.sh | bash
--2025-05-22 19:02:32--  https://raw.githubusercontent.com/Frix-x/klippain-shaketune/main/install.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.108.133, 185.199.109.133, 185.199.110.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.108.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 5197 (5.1K) [text/plain]
Saving to: ‘STDOUT’

-                                                               100%[====================================================================================================================================================>]   5.08K  --.-KB/s    in 0.008s

2025-05-22 19:02:32 (621 KB/s) - written to stdout [5197/5197]

=============================================
- Klippain Shake&Tune module install script -
=============================================

[PRE-CHECK] Klipper service found! Continuing...

libopenblas-dev is already installed
libatlas-base-dev is already installed
[DOWNLOAD] Klippain Shake&Tune module repository already found locally. Continuing...

/home/pi/klippy-env/bin/activate: line 57: PS1: unbound variable
pi@mainsailos:~ $

```

mickeybob00, 2025-05-24:
I am getting the same error

Frix-x, 2025-05-25:
This error happens when your environnement is still running Python 2.x.
Please update your Klipper venv to use Python 3 as this is a requirement for Shake&Tune to work properly. It actually need Python >=3.9 that should be available starting from Debian Bullseye :)

Trailglider, 2025-06-01:
In my case python -V reveals I am running Python 2.7.16. After doing some digging turns out this will require a complete OS reinstall for me as I am on Raspbian 10 (Buster) which only supports Python 3.7. I don't love it..

Strange that mickeybob00 gets same error but has correct Python version.