Frix-x/klippain-shaketune #225: 100% CPU 100% RAM usage leading to system crash during install

southproduct, 2025-07-12:
### K-Shake&Tune module branch

- [x] I confirm using the main branch

### Version

version nothing - couldn't install

### Describe the bug and expected behavior

Raspberry pi model 4

- 1.8 Ghz
- 512 MB swap
- 1GB ram
- 32 GB microSD

**Console**:
```
voron@voron:~ $ wget -O - https://raw.githubusercontent.com/Frix-x/klippain-shaketune/main/install.sh | bash
--2025-07-12 14:41:52--  https://raw.githubusercontent.com/Frix-x/klippain-shaketune/main/install.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 2606:50c0:8001::154, 2606:50c0:8002::154, 2606:50c0:8003::154, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|2606:50c0:8001::154|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 5197 (5.1K) [text/plain]
Saving to: ‘STDOUT’

-                                                 100%[==========================================================================================================>]   5.08K  --.-KB/s    in 0s

2025-07-12 14:41:52 (25.6 MB/s) - written to stdout [5197/5197]

=============================================
- Klippain Shake&Tune module install script -
=============================================

[PRE-CHECK] Klipper service found! Continuing...

libopenblas-dev is already installed
libatlas-base-dev is already installed
[DOWNLOAD] Downloading Klippain Shake&Tune module repository...
Cloning into 'klippain_shaketune'...
remote: Enumerating objects: 1928, done.
remote: Counting objects: 100% (811/811), done.
remote: Compressing objects: 100% (365/365), done.
remote: Total 1928 (delta 605), reused 479 (delta 445), pack-reused 1117 (from 3)
Receiving objects: 100% (1928/1928), 39.19 MiB | 12.87 MiB/s, done.
Resolving deltas: 100% (1211/1211), done.
[DOWNLOAD] Download complete!

[SETUP] Installing/Updating K-Shake&Tune dependencies...
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pip in ./klippy-env/lib/python3.11/site-packages (23.0.1)
Collecting pip
  Downloading pip-25.1.1-py3-none-any.whl (1.8 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.8/1.8 MB 6.6 MB/s eta 0:00:00
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 23.0.1
    Uninstalling pip-23.0.1:
      Successfully uninstalled pip-23.0.1
Successfully installed pip-25.1.1
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting GitPython==3.1.41 (from -r /home/voron/klippain_shaketune/requirements.txt (line 1))
  Downloading https://www.piwheels.org/simple/gitpython/GitPython-3.1.41-py3-none-any.whl (196 kB)
Collecting matplotlib==3.8.2 (from -r /home/voron/klippain_shaketune/requirements.txt (line 2))
  Downloading https://www.piwheels.org/simple/matplotlib/matplotlib-3.8.2-cp311-cp311-linux_armv7l.whl (10.9 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 10.9/10.9 MB 3.1 MB/s eta 0:00:00
Collecting numpy==1.26.2 (from -r /home/voron/klippain_shaketune/requirements.txt (line 3))
  Downloading https://www.piwheels.org/simple/numpy/numpy-1.26.2-cp311-cp311-linux_armv7l.whl (19.5 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 19.5/19.5 MB 8.6 MB/s eta 0:00:00
Collecting scipy==1.11.4 (from -r /home/voron/klippain_shaketune/requirements.txt (line 4))
  Downloading https://www.piwheels.org/simple/scipy/scipy-1.11.4-cp311-cp311-linux_armv7l.whl (65.4 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 65.4/65.4 MB 7.1 MB/s eta 0:00:00
Collecting PyWavelets==1.6.0 (from -r /home/voron/klippain_shaketune/requirements.txt (line 5))
  Downloading https://www.piwheels.org/simple/pywavelets/pywavelets-1.6.0-cp311-cp311-linux_armv7l.whl (4.5 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 4.5/4.5 MB 6.3 MB/s eta 0:00:00
Collecting zstandard==0.23.0 (from -r /home/voron/klippain_shaketune/requirements.txt (line 6))
  Downloading https://www.piwheels.org/simple/zstandard/zstandard-0.23.0-cp311-cp311-linux_armv7l.whl (3.7 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 3.7/3.7 MB 9.8 MB/s eta 0:00:00
Collecting gitdb<5,>=4.0.1 (from GitPython==3.1.41->-r /home/voron/klippain_shaketune/requirements.txt (line 1))
  Downloading https://www.piwheels.org/simple/gitdb/gitdb-4.0.12-py3-none-any.whl (62 kB)
Collecting contourpy>=1.0.1 (from matplotlib==3.8.2->-r /home/voron/klippain_shaketune/requirements.txt (line 2))
  Downloading contourpy-1.3.2.tar.gz (13.5 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 13.5/13.5 MB 13.0 MB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... |

```

Any workarounds other than buying a Raspberry Pi 5 with 32 gb ram max overclocked HD 2000 🤣 ?

southproduct, 2025-07-12:
Solved!

Increased swap from 512 to 8GB and also increased max swap file size to match.

```
CONF_SWAPSIZE=8192
CONF_MAXSWAP=8192
```

Leaving this here for anyone with a similar issue in the future.

eaglerock86, 2025-08-29:
Edit, i just tried that but still didnt work for me :(

eaglerock86, 2025-08-30:
> > > Solved!
> > > Increased swap from 512 to 8GB and also increased max swap file size to match.
> > > ```
> > > CONF_SWAPSIZE=8192
> > > CONF_MAXSWAP=8192
> > > ```
> > > 
> > > 
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > Leaving this here for anyone with a similar issue in the future.
> > 
> > 
> > Hi, I'm having the same issue. should i run those commands before the install ?
> 
> In math you go through a series of steps until you get to a final result... well this is that. It's just the final result. These are not commands you can Copy/Paste.
> 
> I suggest you backup everything to your PC, format the drive, reinstall the OS, and ask an AI model to guide you through the steps required to "Increased swap from 512 to 8GB and also increased max swap file size to match". At least, that's what I did and how I solved the issue.
> 
> Or, you can skip the formatting / reinstall and just ask AI to guide you through the steps required to "Increased swap from 512 to 8GB and also increased max swap file size to match".
> 
> Hope this helps.

Yess, that did do the trick after following how to change the swap size :)
Thank you so much :)

jpbnto, 2025-12-06:
sharing it here incase i will be needing it again

https://forums.raspberrypi.com/viewtopic.php?t=46472

Frix-x, 2025-12-07:
Hi! In the latest version of the develop branch (to be released soon), I removed the dependency on Scipy. So this compilation shouldn't be needed anymore and this problem should be solved automatically. Closing this issue for now, but feel free to reopen if it's not fixed after the release