Frix-x/klippain-shaketune #236: Error while generating the graphs: x and y must have same first dimension

zefir-o, 2025-09-03:
### K-Shake&Tune module branch

- [x] I confirm using the main branch

### Version

v5.2.1-0-g786abbbf

### Describe the bug and expected behavior

No graph when using AXES_SHAPER_CALIBRATION
> Error while generating the graphs: x and y must have same first dimension, but have shapes (85,) and (98,)

### Additional information and klippy.log

[klippy (3).log](https://github.com/user-attachments/files/22113306/klippy.3.log)

loss-and-quick, 2025-09-09:
I have resolved this issue by that:
```diff
--- a/shaketune/graph_creators/plotters/shaper_plotter.py
+++ b/shaketune/graph_creators/plotters/shaper_plotter.py
@@ -11,6 +11,7 @@ from typing import Any, Dict
 
 import matplotlib
 import matplotlib.pyplot as plt
+import numpy as np
 from matplotlib.figure import Figure
 
 from ..base_models import PlotterStrategy
@@ -117,6 +118,17 @@ class ShaperPlotter(PlotterStrategy):
         py = calibration_data.psd_y
         pz = calibration_data.psd_z
 
+        # Ensure all arrays have the same length
+        expected_length = len(freqs)
+        if len(psd) != expected_length:
+            psd = psd[:expected_length] if len(psd) > expected_length else np.pad(psd, (0, expected_length - len(psd)))
+        if len(px) != expected_length:
+            px = px[:expected_length] if len(px) > expected_length else np.pad(px, (0, expected_length - len(px)))
+        if len(py) != expected_length:
+            py = py[:expected_length] if len(py) > expected_length else np.pad(py, (0, expected_length - len(py)))
+        if len(pz) != expected_length:
+            pz = pz[:expected_length] if len(pz) > expected_length else np.pad(pz, (0, expected_length - len(pz)))
+
         # Plot PSDs
         ax.plot(freqs, psd, label='X+Y+Z', color='purple', zorder=5)
         ax.plot(freqs, px, label='X', color='red')
```

zefir-o, 2025-09-09:
Thank you @loss-and-quick for the suggestion. However, I assume there is an error in data collection rather than in the rendering step. Such a fix would mask the issue, potentially leading to incorrect results. 
Would like to hear @Frix-x opinion.

loss-and-quick, 2025-09-09:
> I assume there is an error in data collection rather than in the rendering step. 

I think so too.

steeveone, 2025-12-14:
same error here, using adxl from cartographer
ValueError: x and y must have same first dimension, but have shapes (127,) and (1000,)

marcofailli, 2025-12-23:
News on this? I get the same error while generating the Input shaper graphs

"Error while generating the graphs: x and y must have same first dimension, but have shapes (66,) and (99,)"

joecocio, 2025-12-23:
same problem here as well. fresh os install on the compat-dec2025 branch i get this error, on the main i get the no data error.

Frix-x, 2026-01-04:
This is indeed a known error that I already partially fixed but it was not enough for Klipper users. So I just pushed a final update on the compat-dec2025 branch. So if you are already on this branch, just pull the latest commits and let me know if it's correctly working now!

joecocio, 2026-01-05:
> This is indeed a known error that I already partially fixed but it was not enough for Klipper users. So I just pushed a final update on the compat-dec2025 branch. So if you are already on this branch, just pull the latest commits and let me know if it's correctly working now!

Works for me now thank you for the update!